trezor/trezor-firmware #5045: Change TS3 unknown token/contract address warning screen

Hannsek, 2025-05-15:
Please change the old design:
[image]

to the new one:
[image]
on all affected places

ibz, 2025-05-21:
@Hannsek Are you fine if the title is capitalized (`UNKNOWN CONTRACT ADDRESS`), for consistency with other warnings?

Hannsek, 2025-05-21:
Yes, thanks for this catch.

closed 2025-05-23 by ibz

Ondra-Zik-SL, 2025-06-26:
**QA OK**

**Info:**
 - Suite version: desktop 25.7.0 (ae5b0f63b9337dde09f435baf815de2f192ddf0d)
 - Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteLocal/25.7.0 Chrome/136.0.7103.49 Electron/36.2.0 Safari/537.36
 - OS: Linux x86_64
 - Screen: 2560x1440
 - Device: Trezor T2B1 2.9.0 regular (revision 5e4c97f98d2e139aa82135a0ee43003816f62c6c)
 - Transport: BridgeTransport 3.1.0-bundled.25.7.0