trezor/trezor-firmware #5052: Trezor no longer works with go-ethereum usbwallet library

jtolio, 2025-05-17:
**Describe the bug**

I am trying to understand what has changed in the last year regarding Trezor's interaction with go-ethereum's usbwallet library. This used to work!

As far as I can tell, on both macOS and Linux, when I connect my Trezor (I have tried both a Trezor Model One with the latest firmware and a Trezor Model T with the latest firmware), go-ethereum's USB library, github.com/karalabe/hid, does find a USB entry for the Trezor, but it is for USB Interface 1, not Interface 0. No matter what I do, on Linux and macOS, only the FIDO/U2F interface shows up. However, the Trezor Suite still works, of course.

**Firmware version and revision**
Latest firmwares as of 2025-05-17 for both Model T and Model One

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite
 - OS: macOS, Linux
 - Browser Chrome

**To Reproduce**
Steps to reproduce the behavior:

```
package main

import (
	"fmt"

	"github.com/ethereum/go-ethereum/accounts/usbwallet"
)

func main() {
	hub, err := usbwallet.NewTrezorHubWithWebUSB()
	if err != nil {
		panic(err)
	}
	for _, wallet := range hub.Wallets() {
		fmt.Println(wallet)
	}
}
```

**Expected behavior**
I expect to see hardware wallets! But there are none.

**Additional context**
I understand that this repo is not about the go-ethereum library. I have filed a separate ticket there: https://github.com/ethereum/go-ethereum/issues/31841

This ticket is a hopeful ticket wherein perhaps a Trezor expert will know what changed and will know what advice to provide.

jtolio, 2025-05-17:
Trezor Model T won't let me downgrade back past 2.8.7, and Trezor Model T 2.8.7 has this problem.

prusnak, 2025-05-18:
> go-ethereum's USB library, github.com/karalabe/hid, does find a USB entry for the Trezor, but it is for USB Interface 1, not Interface 0.

This is expected. Interface 0 is a WebUSB interface, not HID interface (which is only used for FIDO).

However, this does not explain why `usbwallet.NewTrezorHubWithWebUSB()` does not work (hopefully that uses WebUSB correctly).

prusnak, 2025-05-18:
I am closing this issue because I want the discussion to only happen in the upstream bug: https://github.com/ethereum/go-ethereum/issues/31841

closed 2025-05-18 by prusnak