trezor/trezor-firmware #5057: Screen artifacts when upgrading firmware of TS3 device

bosomt, 2025-05-19:
**Describe the bug**
Screen artifacts when upgrading firmware of TS3 device
First two screenshots are when upgrading via Trezor Suite
Second two screenshots are when using `trezorctl firmware update`

**Steps to reproduce:**
1. a
2. b
3. c

**Info:**
 - Suite version: desktop 25.5.1 (9a4da1ca88b07ca736c2d7d1ba1d5ae693a07f25)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/25.5.1 Chrome/134.0.6998.178 Electron/35.1.2 Safari/537.36
 - OS: MacIntel
 - Screen: 1470x956
 - Device: 
 - Transport: BridgeTransport 3.0.0-bundled.25.5.1

**Screenshots:**

[image]

[image]

[image]

[image]

prusnak, 2025-05-19:
tangential: showing hex strings with proportional font is **yuck**

Hannsek, 2025-05-19:
We unfortunately don't have monofont in bootloader :/.