trezor/trezor-firmware #5080: pytest retries setting messes up everything except CI runs

matejcik, 2025-05-23:
the global setting for pytest `--retries=5 --retry-delay=1` is messing up all uses of pytest, introducing needless delays for all failures, especially in case the failure takes a long time to hit

it should be isolated to (a) tests against the Trezor emulator (b) CI runs + HW runs, and ideally (c) only some classes of failures -- maybe just errors, not failed assertions?

romanz, 2025-05-23:
Regarding (c), IIUC the retries  currently should be applied only if there are timeouts or `UnexpectedMagic` errors:
https://github.com/trezor/trezor-firmware/blob/8a5f5995228713417a1f007a75291ae37d21d87c/tests/conftest.py#L524-L525

(following https://github.com/str0zzapreti/pytest-retry?tab=readme-ov-file#advanced-options)

Is it OK?

romanz, 2025-05-23:
Would it be OK to remove the following lines from here:
https://github.com/trezor/trezor-firmware/blob/8a5f5995228713417a1f007a75291ae37d21d87c/setup.cfg#L43-L44
and add them to [.github/workflows/core-hw.yml](https://github.com/trezor/trezor-firmware/blob/main/.github/workflows/core-hw.yml)?

Also, I am not sure that we need retries today for emulator tests - they shouldn't be flaky, except for [upgrade tests](https://github.com/trezor/trezor-firmware/issues/4973).

matejcik, 2025-05-23:
> Regarding (c), IIUC the retries currently should be applied only if there are timeouts or `UnexpectedMagic` errors (which may happen due to USB packet loss):
(...)
> Is it OK?

no because you're setting global config for all of pytest, but applying the refinement only in our `tests/` subdir.
there are a bunch of other places where pytest can run, that don't read `tests/conftest.py`

> and add them to [.github/workflows/core-hw.yml](https://github.com/trezor/trezor-firmware/blob/main/.github/workflows/core-hw.yml)?

i would say so, yes

> Also, I am not sure that we need retries today for emulator tests - they shouldn't be flaky

if not, all the better

matejcik, 2025-05-23:
> (following https://github.com/str0zzapreti/pytest-retry?tab=readme-ov-file#advanced-options)

this is not at all a great design, but i guess setting the filtered exceptions this way doesn't hurt anything if the retry options are not applied globally

closed 2025-05-23 by romanz