trezor/trezor-firmware #5120: T1B1 - Adjust the `ButtonRequest` type returned when signing Stellar transactions.

overcat, 2025-05-30:
Currently, when signing Stellar transactions on T1B1, it returns `ButtonRequest.ProtectCall`, which seems inappropriate. Would using `ButtonRequest.ConfirmOutput` and `ButtonRequest.SignTx` be more suitable? Below are the actual logs and my proposed changes. If you agree, I will submit a PR to update it.

```diff
trezorctl -v stellar sign-transaction 'AAAAAgAAAAAsY9w8Ek3jJ8xK8AmIr6hEkqQpo8VO4LhhSC5o00cZKgAAAGQDROWmAAAALAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAVoZWxsbwAAAAAAAAEAAAAAAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAO5rKAAAAAAAAAAAA'
[2025-05-30 15:17:45,673] trezorlib.transport.hid INFO: HID transport is disabled: No module named 'hid'
[2025-05-30 15:17:45,698] trezorlib.transport INFO: Enumerating BridgeTransport: found 1 devices
[2025-05-30 15:17:45,698] trezorlib.transport INFO: Enumerating UdpTransport: found 0 devices
[2025-05-30 15:17:45,700] trezorlib.transport INFO: Enumerating WebUsbTransport: found 0 devices
[2025-05-30 15:17:45,700] trezorlib.client INFO: creating client instance for device: bridge:1
[2025-05-30 15:17:45,703] trezorlib.client DEBUG: sending message: Initialize
Initialize (0 bytes) {
}
[2025-05-30 15:17:45,709] trezorlib.client DEBUG: received message: Features (214 bytes)
[2025-05-30 15:17:45,714] trezorlib.client DEBUG: sending message: StellarSignTx
StellarSignTx (151 bytes) {
    address_n: [2147483692, 2147483796, 2147483648],
    fee: 100,
    memo_text: 'hello',
    memo_type: TEXT (1),
    network_passphrase: 'Public Global Stellar Network ; September 2015',
    num_operations: 1,
    sequence_number: 235565581657440300,
    source_account: 'GAWGHXB4CJG6GJ6MJLYATCFPVBCJFJBJUPCU5YFYMFEC42GTI4MSU2JV',
    timebounds_end: 0,
    timebounds_start: 0,
}
[2025-05-30 15:17:45,718] trezorlib.client DEBUG: received message: ButtonRequest
ButtonRequest (2 bytes) {
-    code: ProtectCall (7),
+    code: ConfirmOutput (3),
}
[2025-05-30 15:17:45,718] trezorlib.client DEBUG: sending message: ButtonAck
ButtonAck (0 bytes) {
}
Please confirm action on your Trezor device.
[2025-05-30 15:17:50,356] trezorlib.client DEBUG: received message: ButtonRequest
ButtonRequest (2 bytes) {
-    code: ProtectCall (7),
+    code: ConfirmOutput (3),
}
[2025-05-30 15:17:50,357] trezorlib.client DEBUG: sending message: ButtonAck
ButtonAck (0 bytes) {
}
[2025-05-30 15:17:51,201] trezorlib.client DEBUG: received message: ButtonRequest
ButtonRequest (2 bytes) {
-    code: ProtectCall (7),
+    code: ConfirmOutput (3),
}
[2025-05-30 15:17:51,201] trezorlib.client DEBUG: sending message: ButtonAck
ButtonAck (0 bytes) {
}
[2025-05-30 15:17:51,675] trezorlib.client DEBUG: received message: StellarTxOpRequest
StellarTxOpRequest (0 bytes) {
}
[2025-05-30 15:17:51,695] trezorlib.client DEBUG: sending message: StellarPaymentOp
StellarPaymentOp (68 bytes) {
    amount: 1000000000,
    asset: StellarAsset (2 bytes) {
        type: NATIVE (0),
    },
    destination_account: 'GAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWHF',
}
[2025-05-30 15:17:51,711] trezorlib.client DEBUG: received message: ButtonRequest
ButtonRequest (2 bytes) {
-    code: ProtectCall (7),
+    code: SignTx (8),
}
[2025-05-30 15:17:51,711] trezorlib.client DEBUG: sending message: ButtonAck
ButtonAck (0 bytes) {
}
[2025-05-30 15:17:52,296] trezorlib.client DEBUG: received message: StellarSignedTx
StellarSignedTx (100 bytes) {
    public_key: 32 bytes 0x2c63dc3c124de327cc4af00988afa84492a429a3c54ee0b861482e68d347192a,
    signature: 64 bytes 0x30776cba15ab1ec7afb13fa390db1525d56e75f32750f66369d64e401bc5d0adf65b8d48a184541417d5dfd3d8c551b77908eb20cf21afe60d4173c204a42b04,
}
[2025-05-30 15:17:52,312] trezorlib.client DEBUG: sending message: EndSession
EndSession (0 bytes) {
}
[2025-05-30 15:17:52,336] trezorlib.client DEBUG: received message: Success
Success (15 bytes) {
    message: 'Session ended',
}
MHdsuhWrHsevsT+jkNsVJdVudfMnUPZjadZOQBvF0K32W41IoYRUFBfV39PYxVG3eQjrIM8hr+YNQXPCBKQrBA==```

overcat, 2025-05-30:
https://github.com/trezor/trezor-suite/issues/19014

matejcik, 2025-05-30:
yup, don't care at all what ButtonRequest codes are sent in the flow, Suite needs should dictate that.
please send the PR. (note that we don't know when we'll be releasing T1B1 firmware next)

overcat, 2025-05-30:
> note that we don't know when we'll be releasing T1B1 firmware next

Okay, I can do some small tricks over at Suite first.