trezor/trezor-firmware #5133: T1 freezes when canceling show public key flow from Suite

Hannsek, 2025-06-03:
**Describe the bug**
When cancelling the Show Public key flow from Suite, T1 starts to flickr.
If the Suite is restarted (`Cmd+R`), it stops flickering, but the device doesn't react to any button press.

https://github.com/user-attachments/assets/b41983bc-3cf2-452f-bb99-12bb17c9d4d7
https://github.com/user-attachments/assets/4728fc6d-6d2a-4408-a708-a37a2e741cb4

**Firmware version and revision**
1.13.1

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite
 - OS: macOS
 - Version: 25.5.2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Account
2. Click on Show public key
3. See error

**Expected behavior**
flow should be cancelled

**Additional context**
I saw this error message in Suite when playing around. Not sure if relevant… 
`Acquire error Initialize failed: assertType: Response of unexpected type: PublicKey. Should be Features, code: Runtime`

romanz, 2025-06-05:
Suite log:
```
2025-06-05T06:54:32.421Z - DEBUG(@trezor/connect): call getPublicKey
DeviceCommands Sending Initialize {
  derive_cardano: false,
  session_id: '8e7a182a95c05f18f018946e8fd7534ba2df8a8499da68232403cc2833748fe6'
}
@trezor/connect handleMessage device-changed
DeviceCommands Received Features (redacted...)
DeviceCommands Sending GetAddress {
  address_n: [ 2147483692, 2147483649, 2147483648, 0, 0 ],
  coin_name: 'Testnet',
  script_type: 'SPENDADDRESS'
}
DeviceCommands Received Address { address: 'mnZqJCnB2VTmj7swejiUYJ4XuddmsKHZeN', mac: null }
Core handleMessage popup-handshake
DeviceCommands Sending GetPublicKey {
  address_n: '(redacted...)',
  coin_name: 'Testnet',
  script_type: 'SPENDWITNESS',
  show_display: true,
  ignore_xpub_magic: undefined,
  ecdsa_curve_name: undefined
}
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
2025-06-05T06:54:32.423Z - DEBUG(request-interceptor): net.Socket.connect - 127.0.0.1:21325
2025-06-05T06:54:37.067Z - DEBUG(request-interceptor): net.connect - 127.0.0.1:21325
2025-06-05T06:54:37.067Z - DEBUG(request-interceptor): net.Socket.connect - 127.0.0.1:21325
2025-06-05T06:54:37.082Z - DEBUG(process-trezord): Checking status (200)
Core handleMessage popup-closed
Core Cleanup...
2025-06-05T06:54:40.432Z - DEBUG(@trezor/connect): call cancel
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
Core handleMessage popup-handshake
DeviceCommands Sending ButtonAck {}
@trezor/connect handleMessage button
@trezor/connect handleMessage ui-button
DeviceCommands Received ButtonRequest { code: 'ButtonRequest_PublicKey', pages: null, name: null }
```

closed 2025-06-18 by romanz

bosomt, 2025-06-18:
**QA OK**

- reload Suite - XPUB dialogue disappears from device display
- cancel dialogue in Suite - XPUB dialogue disappears from device display

**Info:**
 - Suite version: desktop 25.7.0 (4c4289cf9e7e4c3cf217f7974565ed818697bd64)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/25.7.0 Chrome/134.0.6998.178 Electron/35.1.2 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T1B1 1.14.0 bitcoin-only (revision 1aee2ece9b4a2b9952b677d8d8152392e00f91ca)
 - Transport: BridgeTransport 2.0.33