trezor/trezor-firmware #5148: Stellar: if `timebounds` are not set in the tx, then omit displaying it.

overcat, 2025-06-06:
Currently, for a simple send operation, it requires at least seven clicks on Model T. Can we avoid displaying it when the tx does not have timebounds set to reduce the number of clicks needed?

[image]

overcat, 2025-06-06:
If approved, I will submit a PR. 

I haven't submitted a PR here in a long time, so I want to start with a small PR to refresh my memory of the complete process.

Hannsek, 2025-06-06:
Hello, thanks for this. Let's narrow the UX. I am not really familiar with what needs to be displayed or not in the Stelar ecosystem. The basic send flow on the device should follow this structure:

`Recipient address`
`Amount`
`Summary - total amount + fee`

[image]

All other details (fee info, account info, blockhash (for Solana), …) should go under the on-device `menu`:

[image]

[image]

I am happy to discuss details of Stellar on a call so it is clear for both sides.

overcat, 2025-06-06:
Hi @Hannsek, thank you for this information. I will take some time to see if I can align Stellar's process as closely as possible with the one in the image above.

Hannsek, 2025-06-06:
Ok, thank you. Can you tell what are all the information needed to be displayed on Trezor screen?

overcat, 2025-06-09:
> Is this really necessary to display it? What value it brings to the users? 

I believe that in most cases, it is not important for the user. However, if the user wants a transaction to be submitted to the network and take effect only after (or before) a specific point in time, then it becomes very important for the user.

> How it would be displayed?

You can check the 19-second mark of the first video, where it shows the timebounds; I think it can be placed there.

Hannsek, 2025-06-09:
How often is `memo` used when doing standard operations (not sending to exchange)? I am wondering if we should use _hard_ or _soft_ warning if memo is not set.

Hannsek, 2025-06-09:
What is under "verify user" item in the menu?

Hannsek, 2025-06-16:
Thanks for the clarification. Here is my proposal for the standard `send flow`:

[image]

And a proposal for `Add trust` flow:

[image]

Please take a look and tell me if you think it is wrong.

Hannsek, 2025-06-16:
Thanks for that 🙏, I redone it a bit. Please take a look. Send flow:

[image]

Create trustline flow:

[image]

> Should we prompt the user if they are not using the public network? If the user is testing on a test network, this could be useful as they can confirm that they are signing transactions on the test network.

I wouldn't do it as it cannot harm anyone when sending transaction on testnet instead of mainnet.

overcat, 2025-06-16:
Hi @Hannsek,

The design looks great! I do have a couple of suggestions, though.

First, I think `Signing with` would be a better phrase to use here. Also, if `tx.source_account` isn't `trezor_account`, the assets should be spent from `tx.source_account` instead of `trezor_account`. This is because Stellar supports multisignature, and `trezor_account` would be set as one of the signers for `tx.source_account`.

[image]

Second, when `tx.source_account` isn't `trezor_account`, we should display a prompt on the home screen. I've actually included this in the code already, you can see it here: https://github.com/trezor/trezor-firmware/pull/5159/files#diff-0cb8d273e306cd5a00e634af5675de04f84c4a54d6bf25779be3e9b62e852e92R14-R20

I'm currently downloading the Nix dependencies on my end, which might take a little while. But I can definitely send over some screenshot later if you need them!

overcat, 2025-06-16:
Hi @Hannsek, screenshot ready.
[image]

Hannsek, 2025-06-16:
> First, I think Signing with would be a better phrase to use here. 

I kinda agree, however we use "Send from" for all coins. Also from translation POV it would be unnecessary burden.

> Also, if tx.source_account isn't trezor_account, the assets should be spent from tx.source_account instead of trezor_account. This is because Stellar supports multisignature, and trezor_account would be set as one of the signers for tx.source_account.

Can you explain this a little bit more please? Does that mean that if multisig is used, transaction might not be sent from account from Trezor but from outside? Thus we don't know the derivation path and we cannot display it. So we display only the address itself?

Hannsek, 2025-06-16:
If the above is valid. I proposed a slight design change.

1. if the source account is not owned by Trezor, display waring in the beginning of the flow.
2. if the source account is not owned by Trezor, adjust "Send from" screen by removing derivation path and add "externally owned" as a value to "Account" key

[image]

overcat, 2025-06-16:
Hi @Hannsek 

My apologies if my previous explanation wasn't clear enough. Let me clarify with an example.

Let's imagine we have an **external** account, `GA`, and a **Trezor** account, `GB`. The user has set up a multi-signature requirement for `GA`, meaning any transaction initiated from `GA` must be signed by **both** `GA` and `GB` to become valid.

Now, the user constructs a transaction. In this tx, `GA` is set as the source account, and it's sending 100 USDC to another account. This transaction is then sent to the Trezor device. At this point, the Trezor needs to review and sign it.

Once tx has been signed by both `GA` and `GB`, it can be broadcast to the network. The 100 USDC will then be debited from GA's account.

This is why saying `Send from` isn't quite accurate in this context. While the Trezor account (`GB`) signs the transaction, the USDC is actually debited from the `GA` account. `GB` is just one of the required signatories, not the source of the funds.

overcat, 2025-06-16:
Also, in other scenarios, like with [manage data operations](https://developers.stellar.org/docs/data/apis/horizon/api-reference/resources/operations/object/manage-data), we're not actually sending anything; we're just setting a data entry. I feel `Signing with` would better suit all these different situations.

Hannsek, 2025-06-16:
@overcat Please let me know if my proposal is fine from your side.

overcat, 2025-06-16:
Hi @Hannsek, fantastic work! I can get a PR going for this, if you'd like?

Hannsek, 2025-06-16:
That would be great and very much welcomed. Thank you!

overcat, 2025-06-17:
Hi @Hannsek 
Do we need to design a UI for the [Create Account Operation](https://developers.stellar.org/docs/build/guides/basics/create-account#create-account) as well? It's quite similar to the Payment Operation, but its purpose is to send XLM (specifically XLM, and the amount must be 1 or greater) to a new Stellar account. This action activates the account, and the XLM is then added to the new account's balance.

Since it's essentially just sending XLM to another account, I think we can reuse most of the Payment Operation's UI. Perhaps we could just add a small hint or message indicating that the funds are being sent to a new account?

[image]
[image]

[image]
[image]

Hannsek, 2025-06-17:
Do we even need to tell it to user? Can account creation look the same as standard send flow? Or is it important for the user to know about account creation?

Hannsek, 2025-06-30:
Thank you very much. This is great. 🙏

I have few nits that should be changed:

[image]
A menu needs to be introduced here as user must have a way how to cancel the whole flow here.

[image]
Header should be changed here:
Address
Recipient #1

[image]

- Header should be changed here:
Amount
Recipient #1
- just "Tap to continue" in the menu. No "confirm" here.

[image]
There should be no "tap to continue" in the footer. User should be able to go just back with X.

overcat, 2025-07-01:
Hi @Hannsek, 

> A menu needs to be introduced here as user must have a way how to cancel the whole flow here.

It looks like the `show_warning` interface on T3B1/T3T1 was intentionally designed to **not allow** interrupting the flow on that screen. I've linked the relevant code below for reference:

- https://github.com/trezor/trezor-firmware/blob/367bf53ff2f718f1ab3f420e889075ce1d7024d7/core/embed/rust/src/ui/layout_delizia/ui_firmware.rs#L1178
- https://github.com/trezor/trezor-firmware/blob/367bf53ff2f718f1ab3f420e889075ce1d7024d7/core/embed/rust/src/ui/layout_caesar/ui_firmware.rs#L1290

If you're sure you'd like to change this, I can implement it in a separate PR, as it would affect all interfaces that use this function.

---

I've made the other changes as you suggested, with one minor adjustment: the display shows "More info" (at 16s). This is hardcoded within `trezor_ui`. If we need to change it, I can address it in a small, separate PR.

https://github.com/user-attachments/assets/8f5f7e35-82bd-439e-8c75-5ff84ba011a5

---

Also, since we've introduced `Recipient #{output_index}`, I'm planning to add this information to other send operations for consistency. Currently, operations that allow sending tokens include:

- `Payment`
- `Create Account`
- `Merge Account`: Send All XLM to Destination Account
- `Path Payment Strict Receive`
- `Path Payment Strict Send`

For the `Merge Account` operation, I can easily add this to the existing interface. 
However, for `Path Payment Strict Receive` and `Path Payment Strict Send`, I think we should redesign the UI.

Let me briefly introduce these two operations:

`Path Payment Strict Receive`

```
required StellarAsset send_asset = 2;         // asset we pay with
required sint64 send_max = 3;                 // the maximum amount of sendAsset to send (excluding fees), the actual amount may be lower than this
required string destination_account = 4;      // recipient of the payment
required StellarAsset destination_asset = 5;  // what they end up with
required sint64 destination_amount = 6;       // amount they end up with, the amount received by the recipient must be equal to this value
```
`Path Payment Strict Send`
```
required StellarAsset send_asset = 2;         // asset we pay with
required sint64 send_amount = 3;              // amount of sendAsset to send (excluding fees), the amount sent by the sender must be equal to this value
required string destination_account = 4;      // recipient of the payment
required StellarAsset destination_asset = 5;  // what they end up with
required sint64 destination_min = 6;          // the minimum amount of dest asset to be received, the actual amount may be higher than this
```
If you need a more detailed explanation, you can take a look at [this article](https://medium.com/stellar-community/understanding-stellar-path-payments-5eefe55b071b).

overcat, 2025-07-03:
My apologies for not being clear! I've recorded a video demonstrating 5 operations:
- Create Account (5s)
- Payment (8s)
- Path Payment Strict Send (10s)
- Path Payment Strict Receive (18s)
- Account Merge (27s)

You'll notice that for `Path Payment Strict Send` and `Path Payment Strict Receive`, the `Recipient #{output_index}` field is missing when compared to the UI for other operations. Also, I'm wondering if the display for the `issuer` field should be redesigned? Just like what we did for the `Payment` operation.

https://github.com/user-attachments/assets/b5ecbf4d-8a25-4e0f-a79b-2b6d3e3156a2