trezor/trezor-firmware #5204: T1B1 currency symbol is missing.

Ondra-Zik-SL, 2025-06-16:
**Describe the bug**
If the amount string is too long, the currency symbol doesn't appear. Please check the formatting to ensure the symbol is always shown, even with longer values.

[image]

romanz, 2025-12-04:
@Hannsek Would it be OK to move the symbols to the next line (so they won't be hidden in case of a large amount)?

For example:
```
Send
AAAAAAAAAAAAAAAAAAAAAAAAA
ETH to 0xBBBBBBBBBBBBBBBB
BBBBBBBBBBBBBBBBBBBBBBBBB
BBBBBBBBBBBBBBBBBBBBBBB ?
```

```
Really send
AAAAAAAAAAAAAAAAAAAAAAAAA
ETH, paying up to 
CCCCCCCCCCCCCCCCCCCCCCCCC
Wei for gas?
```

Hannsek, 2025-12-07:
I wouldn't change the how the information architecture. So I'm for

```
Really send
AAAAAAAAAAAAAAAAAAAAAAAAA
ETH (ARB), paying up to 
CCCCCCCCCCCCCCCCCCCCCCCCC
Wei for gas?
```

closed 2025-12-11 by romanz