trezor/trezor-firmware #5294: Add support for EthereumSignTypedHash to new firmware

mdehoog, 2025-07-03:
The [Base](https://github.com/base) and [Optimism](https://github.com/ethereum-optimism) teams use security councils for contract upgrades. These security councils are made up of globally distributed individuals who maintain a hardware key as a signer to an onchain multisig contract (we use [Safe](https://github.com/safe-global/safe-smart-account)).

Generally these individual signers use Ledger for their hardware wallets. However, recently we'd like to add Trezor support to our workflow. The contract upgrade workflow is to open a PR against [ethereum-optimism/superchain-ops](https://github.com/ethereum-optimism/superchain-ops/) or [base/contract-deployments](https://github.com/base/contract-deployments/) with the upgrade script as well as validation instructions.

You can see an example of one of these validation documents [here](https://github.com/base/contract-deployments/blob/main/mainnet/2025-05-15-eip1559-denominator-reduction/VALIDATION.md). Note in particular, the most important piece of the validation is checking the EIP-712 message hash.

Here is a video of a Ledger device confirming a typed EIP-712 payload:
https://github.com/user-attachments/assets/aa4fa487-0699-4399-bda6-bf1f849ea226. Note that it doesn't support showing the individual values in the EIP-712 message struct; rather it hashes the entire message and shows the message hash.

Similarly, the Trezor Model One only supports `EthereumSignTypedHash` and not `EthereumSignTypedData` (added in #1970). You can view the confirmation of one of these messages in this video: https://github.com/user-attachments/assets/4e0519ad-fec5-4fa7-aeb4-1ecd39f7d2e9.

However, the Trezor Model T / Safe 3 / Safe 5 don't support `EthereumSignTypedHash`, they only support `EthereumSignTypedData`. This forces us to send the entire typed payload to the device, rather than just the hash. Here's a video of what that looks like: https://github.com/user-attachments/assets/dc3e42cf-0d33-4b69-96fd-555d7df96de2. Note that the user has to confirm each of the fields in the struct, including the `data` field.

Given our requirement for the user to confirm the message hash matches exactly, this doesn't work for these Trezor devices. We could totally update our process to print out the full typed data struct and get the user to validate this, however this is both laborious (there's a bunch of fields in Safe transactions that are irrelevant) and error prone (the data field can be very large as it is calldata, expecting a user to validate every byte is tricky; not to mention that a single byte change can completely change the behavior of the upgrade). This is what hashes are for: any small change in the data will cause the hash to change completely, so validating the hash is much safer.

We'd love to see newer Trezor devices support `EthereumSignTypedHash`. I believe the intent for `EthereumSignTypedData` is to validate user-readable structs (the example given in [EIP-712](https://eips.ethereum.org/EIPS/eip-712) is an email with `from/to/contents` fields), and it doesn't make sense for signing transaction calldata. Signing the hash directly is much safer for this usecase.

matejcik, 2025-07-03:
We don't want to add the option to send just the hash to Trezor -- it is **in no way** safe, or "safer" than something. The actually safe way to implement this is to decode calldata embedded in the EIP712 message.
(which is on our medium-term roadmap, but no ETA because there's a _lot_ of competing priorities on there)

(even better: don't stuff calldata in there in the first place. make it structured so that users **can** validate it, and make the relevant smart contracts accept structured messages.)

That said, I believe your usecase would be covered by adding an option to **display the hashes** to the existing `EthereumSignTypedData` feature. Then, in the summary menu before signing, you'd have the option to review the hash and compare to what your website tells you.

mdehoog, 2025-07-03:
Thanks for the thoughtful reply, you raise some great points.

>The actually safe way to implement this is to decode calldata embedded in the EIP712 message.

That's a fair statement 👍 . In our case, we use a double or triple nested Safe as our multisig, and the signers with the hardware keys are on the leafs of the multisig structure, so in the end they are signing hashed approvals anyway; decoding the calldata would just reveal a call to `approveHash(bytes32)` which is not more descriptive to the signer than the outer message hash.

It wouldn't be possible for a hardware device to display the full context of the hash behind signed without passing it a lot of contextual information as well. We achieve this validation in our terminal scripts instead... the signature on the device is the final step in the signing process, and the only thing that signers care about is the message hash.

>(even better: don't stuff calldata in there in the first place. make it structured so that users can validate it, and make the relevant smart contracts accept structured messages.)

This is also a good point. We are pretty tied to Safe as our multisig contract, given their great history of safety and audits, and much of our tooling assumes Safes. Their EIP-712 data structure looks like [this](https://github.com/safe-global/safe-smart-account/blob/4be4e95e86e1de5f0592aae7879edaa6f455148f/contracts/Safe.sol#L449-L459). It's pretty much a non-starter to switch our multisig implementation to something a little more friendly to human-understandable transaction data.

>That said, I believe your usecase would be covered by adding an option to display the hashes to the existing EthereumSignTypedData feature. Then, in the summary menu before signing, you'd have the option to review the hash and compare to what your website tells you.

This would be a workable solution, and happy to re-implement #5286 in this way if you believe adding `EthereumSignTypedHash` support is a no-go. I would say that it's going to require some significant changes to our signing flow (we need to expose the underlying typed data struct to the signing hardware rather than the hash, and add support to our signing script for passing this data to the hardware devices), but it's doable. I would just point out that, for us, it's no safer, given the structured data is just as obfuscated as the hash itself, so we'd be telling signers to ignore everything but the message hash. Overall it would be easier to instead support `EthereumSignTypedHash` (happy to prefix the signing flow with a big warning like you do in the legacy firmware), but your reluctance is understandable 👍 .

mdehoog, 2025-07-10:
Hey @matejcik, thanks for the proposal. I've chatted to @mds1 and we're happy with this approach. We can make changes to our scripts to expose the underlying EIP-712 typed data structure, so the first drawback is solvable.

For the second, I have a slight preference for providing the `message_hash` as an argument, rather than doing 2 rounds with the device. This has the added advantage of actually validating that the expected message hash being showed on the host app is the same hash calculated and signed by the device.

closed 2025-07-19 by obrusvit