trezor/trezor-firmware #5470: Trezor 5: RSOD Internal Error (UF)

Midar, 2025-08-03:
**Describe the bug**
I just randomly got an RSOD that told me to go to trezor.io/rsod. However, that error is not listed there. It just says:

```
Internal Error
        (UF)
```
And that's it.

**Firmware version and revision**
2.8.10

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: None, was using U2F
 - OS: Linux
 - Browser: None, was using U2F for SSH
 - Version: OpenSSH_9.9p1

**To Reproduce**
I don't have a reproducer, but getting random errors from a HSM that are not documented and don't say what they are are super concerning.

**Expected behavior**
At least instructions on trezor.io/rsod that explain what "(UF)" means? The best guess I have is "Use after free"?

**Screenshots**
-

**Additional context**
I don't expect a fix since there is not much to go by, but at least document what this is and how concerning this is? At the very least, state whether this could have lead to corruption of internal data, could have caused leakage of keys, etc.

matejcik, 2025-08-04:
UF is an "Usage Fault" meaning most likely either (a) illegal instruction (which is typically a Rust panic), (b) or unaligned memory acces.

If it is a Rust panic, it most likely comes from the UI code, meaning we're trying to render something unexpected. You may even be able to reproduce the problem -- did you do some sort of UI interaction? Click, swipe, etc.?

>  state whether this could have lead to corruption of internal data, could have caused leakage of keys, etc.

The fact that you see a RSOD typically means that some sort of internal integrity check was triggered. In other words, the defences are working as expected, and your data and keys are not in danger.

Hannsek, 2025-08-12:
@Midar If it is reproducible, please let us know. We'll try to fix it.

Also, we'll add it to KB article. 

Closing this issue for now.

closed 2025-08-12 by Hannsek