trezor/trezor-firmware #5487: UI: After successful pairing, user is returned to “Pair new device” screen instead of homescreen

Hannsek, 2025-08-05:
When the device is already connected to a computer and the user initiates “Pair with new device” from another client (e.g., mobile), the flow completes successfully. However, instead of proceeding to the homescreen, user is brought back to the “Pair new device” screen.

Fix post-pairing navigation to continue to the homescreen.

romanz, 2025-09-03:
Does it reproduce after https://github.com/trezor/trezor-firmware/pull/5592?
(IIUC, it prevent the device from connecting to multple hosts)

romanz, 2025-09-07:
Tested locally:
- paired with a T3W1 already connected to another host
- after pairing, the device menu is closed and the homescreen is shown.

closed 2025-09-07 by romanz